BeyondRobotix/mavesp8266 #5: Set up Travis build

LorenzMeier, 2016-01-16:
@dogmaphobic Volunteering to help with that.

dogmaphobic, 2016-02-01:
The build itself is setup and working. I just need to figure out where to upload the binaries.

dogmaphobic, 2016-02-01:
https://travis-ci.org/dogmaphobic/mavesp8266

ivankravets, 2016-05-13:
@dogmaphobic do you need any help to integrate this repo with Travis?

dogmaphobic, 2016-05-13:
It's already there, thanks! :) At least the build. What's missing is an automated way to make the distribution available. I also need the *Build Number* to come off Travis instead of having to manually edit it each time.

ivankravets, 2016-05-13:
>  What's missing is an automated way to make the distribution available. 

PlatformIO allows uploading of existing firmware without project rebuilding and project source code. 
1. Place firmware to `.pioenvs/esp12e/firmware.bin`
2. Upload it using `platformio run -t uploadlazy -e esp12e --disable-auto-clean`.

In this case, PlatformIO will install only the upload tools without toolchains, frameworks, SDKs and etc.

> I also need the Build Number to come off Travis instead of having to manually edit it each time.

Could you explain in details?

dogmaphobic, 2016-05-13:
What I meant was to find a place where to put the resulting binary so people can download it. Right now I'm manually building and uploading to my own server and updating the link here:

https://pixhawk.org/peripherals/8266

Regarding automated build numbering, I just need to write a script that reads the current *Build Number* off some text file, increments it, and saves it back (to the repository). It would then generate a header file with the build number instead of it being hard coded here:

https://github.com/dogmaphobic/mavesp8266/blob/master/src/mavesp8266.h#L64-L67

P.S. For what t's worth, for distribution, I use `-e esp01_1m`

ivankravets, 2016-05-13:
PlatformIO is very powerful, flexible and easy for use Build System. Also, Travis provides [environment variables](https://docs.travis-ci.com/user/environment-variables/#Default-Environment-Variables). You need `TRAVIS_BUILD_ID`.

Why do you need this extra script? https://github.com/dogmaphobic/mavesp8266/blob/master/esp_extra.py . You can use [upload_resetmethod](http://docs.platformio.org/en/latest/projectconf.html#upload-resetmethod) instead overriding.

The final `platformio.ini` without extra scripts
```ini
[env:esp01]
platform = espressif
framework = arduino
board = esp01
upload_speed = 921600
upload_resetmethod = nodemcu
build_flags = !echo "-DBUILD_ID="$TRAVIS_BUILD_ID
```

Then, you can upload firmware to your own server or, for example, to @bintray. See https://docs.travis-ci.com/user/deployment/

dogmaphobic, 2016-05-13:
> PlatformIO is very powerful, flexible and easy for use Build System.

Believe me, I thank PlatformIO every day :) I've been writing my own mess of Makefiles handling all these different platforms for years. When I first heard of PlatformIO it was a dream come true.

> You need TRAVIS_BUILD_ID.

Yes, I have not spent any time dealing with this. I figured Travis would have something but I was worried it would be some large (> 16 bits) number. I currently use a 8+8+16 scheme (into a 32-bit number) to save the full version. While googling that, I found the page with all the variables. `TRAVIS_BUILD_NUMBER` might be something to look too. As I said, I have not had the time to look at any of this yet.

> Why do you need this extra script? https://github.com/dogmaphobic/mavesp8266/blob/master/esp_extra.py . You can use upload_resetmethod instead overriding.

That came about because at the time, by default PlatformIO would not properly reset the ESP12 I was using. A quick search on google pointed to that solution, which I used and it just worked. That all happened within a couple of hours of first hearing about PlatformIO and I have not spent much time thinking about it since.

> Then, you can upload firmware to your own server or, for example, to @bintray. See https://docs.travis-ci.com/user/deployment/

My own server is actually *my own, personal server*, not some general hosting service. I would need to setup an account with some place so Travis could upload the binary automatically. I cannot allow Travis to upload anything to my own sever. Again, I have not had the time to think about any of this.

Thanks!

ivankravets, 2016-05-13:
Yes, `TRAVIS_BUILD_NUMBER` is better for you. The last value is `45`.

In any case, thanks a lot you that use PlatformIO! Don't hesitate to contact us for the other questions.

dogmaphobic, 2016-05-13:
Ivan, thank you for putting together such a polished system. I really like what you guys built. Everything "just works"!