BeyondRobotix/mavesp8266 #7: Volunteer for helping in optimize code for pixhak platform

gianni-carbone, 2016-01-27:
Hi,
  I've had to carry some change to code to let it work on Pixhawk Autopilot (HK32 clone)
Now it's working pretty fine.
I'm using ESP8266-07 module and I've notice seriously problems with softAP in WPA2-WEK mode, so I switch in open AP mode

I'm offering volunteer for code optimization and improvement.
Let I know.
Bye,
G. Carbone

dogmaphobic, 2016-01-27:
As long as using open AP is an option, it should be fine. Please submit a PR so I can see the changes you've made.

Thanks!

gianni-carbone, 2016-01-27:
I suggest to start in open AP mode by default and then allow to switch in WPA/WEP mode.
The major change I made was:
start in open AP mode, using 57600 default UART speed, using 14551 default UAV udp port and start with no remote broadcast address  but 192.168.4.2 that is the first dhcp address released to gcs client.
I'm at working now I will send you the code as soon as possible.
GC

dogmaphobic, 2016-01-27:
Ok, so we need an option for enabling/disabling broadcast. Everything else (UDP port, UART speed, etc.) can already be done using the current set of options (parameters). If you set the password to blank (""), it runs as an open AP.
> 192.168.4.2 that is the first dhcp address released to gcs client

There is no guarantee that will always be the case. You cannot make that assumption.

I'm not sure I understand why you made any of those changes. What are you using for GCS? Most of those things can be done/changed using existing MavLink messages (parameters).

gianni-carbone, 2016-01-27:
The cgs I'm using is qgroundcontrol and it was built compiling code about two weeks ago. My workstation runs Ubuntu 12.04. I tried to use ubuntu 14 but there were problems during compilation. 12.04 worked fine.
Surely I've a up to date qcgs. Infact the wifi module is now shown as #204 in the parameters pane  and an alert says me I've to quit cgs and update the firmware.At the same time the workstation seems to have problem with broadcasts message.
By the way, using the above default parameters it let me connect to the uav with no problems just plugging mains and connect to wifi ap. 
That was not possible with previous default configuration.
Naturally I'm going to update firmware and qground control cgs...
Thank you for reply,
GC

gianni-carbone, 2016-01-27:
Hi,
with update px4 firmware and qgroundcontrol your original code work very good.
Thanks

dogmaphobic, 2016-02-01:
> with update px4 firmware and qgroundcontrol your original code work very good.

Perfect. Thanks!

closed 2016-02-01 by dogmaphobic