BeyondRobotix/mavesp8266 #16: Dual SSID/Password

dogmaphobic, 2016-04-30:


mzahana, 2016-04-30:
I tried to use the STA mode by changing the default ssid and password to my router's, and changing the default mode to STA. Then, compiled the code and uploaded it to eps. It didn't work. When I use the normal defaults, AP mode, it works. Any ideas on how to solve this?
Thanks.

dogmaphobic, 2016-04-30:
I just left a comment on gitter:

*I’m still tweaking it. That’s why have not let it loose in the wild. I have not seen the problem you mention but there are a few other things I want done before calling it ready. For instance, you definitelly need two sets of SSID and Password. One for AP and another for STA. When in STA mode and it can’t connect, it automatically reverts to AP mode, currently using the same SSID and Password. Not a good idea if it matches an existing AP.*

I'm almost done with the second set of SSID/PWD and I also have a new version of QGC supporting it. It will all be done soon.

mzahana, 2016-04-30:
Thanks for the effort. I am waiting to try it out. The QGC will be updated in the daily build?

dogmaphobic, 2016-04-30:
There is one PR in already but I have not merged it. I'll wait until I have the additions to MavEsp done before committing it all.

dennisss, 2016-04-30:
@mzahana Another thing to note is that changing the default parameters will not necessarily take effect upon upload as the esp will try to first use any settings stored in flash memory from the previous run. If you are trying to modify the parameters, change them after uploading in QGC. QGC should allow you to access all options in the Parameters area regardless of the status of the WiFi Bridge gui

mzahana, 2016-04-30:
Can I do that on Pixhawk? I am not using Pixracer.

mzahana, 2016-04-30:
Or you mean that I use the PixRacer AP first to set the ESP parameters, then switch to STA mode?

dennisss, 2016-04-30:
Yeah, I've only tested on the PixRacer, but currently I think the parameters are only broadcast/settable if you are directly connected to the ESP via WiFi. QGC should auto-connect to it on the same network and then you can do that.

dogmaphobic, 2016-04-30:
> Can I do that on Pixhawk? I am not using Pixracer.

Yes, I'm using the Pixhawk for all this. I built a little ESP-07 with an external antenna for it.

mzahana, 2016-04-30:
Nice. Can you share your ESP-05 setup, and where i can get the parts?

Thanks.

dogmaphobic, 2016-04-30:
I took a picture of the *before* with the idea of putting together a wiki but never got around to it. I got the ESP-05 off eBay. The rest is just stuff I already had (thus the huge connector). The DC-DC step down is to bring the Pixhawk's 5V to the ESP's 3.3V.

[image]

dogmaphobic, 2016-05-01:
Done in #17.

closed 2016-05-01 by dogmaphobic

Puav, 2016-12-13:
Hallo guys,
and thanks for the great work!
I'm trying to use the MAVESP in station mode with different router (even mobile phone AP) but with no success.
The best situation I experienced is: MAVESP try to connect to the router (i got short notification fro the router) and then the connection quickly drop; after about 1 min sometimes MAVESP switch back to AP mode (sometimes does nothing and need to reboot).
Thanks a lot to anybody in advance!
Any help will be greatly appreciated!

Puav, 2016-12-13:
Hey guys,

after thousands attempt, i finally find what could be the problem.
I succeded to connect to my router and my mobile phone AP just using PixRacer as SSID and pixracer as password. Any other name, password and any possible wifi configuration will not work, then i assume i make some mistake with HTTP command line to set SSID and password.

Any help or suggestion is still very appreciated. 
Thanks a lot

dennisss, 2016-12-13:
I've had success with `http://192.168.4.1/setparameters?mode=1&ssidsta=networkname&pwdsta=thepassword` with no cutouts at all.

If your computer is on the same network as the ESP, then you should use QGroundControl to configure the parameters just to be safe (it has a separate GUI tab for the wifi bridge when connected directly via wifi).

Also do note that the maximum supported length of the password or ssid is 16 characters.

Puav, 2016-12-13:
Thank you Dennis,

i tryied many times with this http address, bt never worked. After many attempts i succeded with single parameter (first SSID, then password and at the end mode=1). I still face some limitations like password lenght; in my case a password longer than 8 characters doesn't work.
I even tryied many times to configure using QGroundControl but i cannot even enable station mode.