BeyondRobotix/mavesp8266 #22: How does the .pioenvs file generate, especially  , there is LWIP file .

liangzelang, 2016-05-10:
follow the project guide，  I find one file .pioenvs  come into being ,and i have some questions:
 1, how does the   XXX.a  files  generate
2,   how does the LWIP file generate ,because  i  don't find  this file before compile . 

thank you .

dogmaphobic, 2016-05-10:
Sorry, but I don't think I understand the question. Those *.a files are libraries, which are then linked to the final executable. The firmware image in this case. Platformio, along with the ESP8266/Arduino tool set automatically detect the various "libraries" in use based on the preprocessor finding the header files within the source files. It then compiles the library sources. The result of this process are the libraries (the *.a) files, which are then linked to the executable.

What do you mean by "LWIP" file?

liangzelang, 2016-05-10:
@dogmaphobic    you look at   .pioenvs/esp12e/ESP8266WiFi  ,in this folder ,there will be  three folder  ：include，arch ，and  lwip  。and I don't know how does this file generate。

liangzelang, 2016-05-10:
sorry to bother you again, what is the relationship between this code and the code of  esp8266/arduino (https://github.com/esp8266/Arduino)   .
can I think that  the code of esp8266/arduino is the arduino core ,including lots of libraries  ,and this project will using the arduino core by update the arduino code when compile?

ivankravets, 2016-05-13:
@liangzelang the source code of Arduino for ESP8266 is located in `~/.platformio/packages/framework-arduinoespressif`

liangzelang, 2016-05-13:
@ivankravets    I got it ,thank you very much !!!

closed 2016-05-13 by liangzelang