BeyondRobotix/mavesp8266 #23: where is the makefile of this Project?

liangzelang, 2016-05-10:
https://github.com/plerup/makeEspArduino
above address,  it provides the makefile of one esp8266 arduino project  .And i can't find this file in your project.  could you please tell me where is the makefile of the project.

ivankravets, 2016-05-13:
@liangzelang do you need any help here?

liangzelang, 2016-05-13:
@ivankravets  Actually,Not fully understand this question . I don't know  whether it‘s right or not : We can configure this project  in the platformio.ini file (set the home_dir, envs_dir ,external lib,platform ,framework and so on) ,   and especially we can allow to launch extra script using SCons tool. in  the esp_extra.py we pass the changes (we make these changes in the platformio.ini)  into defaultenvironment (this defaultenvironment contains toolchain path ,SDK path , libraray path,src path etc) .So when we type the command "platformio run" ,  the Scons tool get this command and Take appropriate action.   Am I right???   hope you can understand what i said ,because my English is not good.

ivankravets, 2016-05-13:
> So when we type the command "platformio run" , the Scons tool get this command and Take appropriate action.

Scons is build tool, like API. When you type `platformio run`, PlatformIO process a lot of different steps:
1. Install dependent packages/software per platform/board
2. Configure own Build Environment based on SCons API, platform, framework and board settings
3. Ask SCons to process configured environment.

SCons like a Make utility but written in Python and cross-platform.

liangzelang, 2016-05-14:
@ivankravets   yeah，  I got it . thank u very much.

closed 2016-05-14 by liangzelang