BeyondRobotix/mavesp8266 #25: Trouble getting Parameters

kabauterman, 2016-06-11:
Hello and Thank you for the great work, spme time ago I did something like this with a wifi serial brige project for the ESP8266 but had no real luck you project seems to be great for what I wanted :)

I had it running for some days, with a good reach and averything working fine, but for now I can connect, I get battery, GPS and so on, but mission planner and tower stuck at getting the parameters.

When I look at the status page there are a lot of lost packages I do not know where to start to find out whats wrong.

I did reset it and flashed it again but no luck so far.

[image]

skorokithakis, 2021-10-02:
I'm having the same issue, the artificial horizon/modes/etc work okay in the GCS, but it won't receive parameters.

skorokithakis, 2021-10-02:
Turns out I needed to set:

```
parachute set SERIAL6_PROTOCOL=2 SERIAL6_OPTIONS=0 SERIAL6_BAUD=921
```