BeyondRobotix/mavesp8266 #30: password change

dronophobia, 2016-08-16:
Hi, it's impossible for me to change the password with qgroundcontrol and the web interface. Is "pixracer" password hard-coded?

dogmaphobic, 2016-08-16:
Yes, the current web interface is still a bit *bare bones* and not truly conducive to changing password or SSID through it. The strings are encoded into 4 32-bit values as the MavLink protocol only supports numbers (not strings).

With that said, you can use QGC and its WiFi panel to change the password and/or the SSID. The default password is `pixracer` but it can be changed to anything from up to 32 characters.

lawnmower4x, 2016-08-21:
The password can not change with qgc 3.0.0 (mac) and /or ios beta and/or the webinterface. Is it possible in another way? 
I don't want to fly with password " pixracer" because i am not the only one with a pixracer here.

dogmaphobic, 2016-08-22:
I've moved this to QGroundControl as it seems to be an issue with it not showing the interface for changing the WiFi Bridge settings:
https://github.com/mavlink/qgroundcontrol/issues/3968

closed 2016-08-22 by dogmaphobic