BeyondRobotix/mavesp8266 #33: How To Query Version Number

jpaulos, 2016-10-17:
How can I query the version number when connected over Wifi?  The setup instructions at https://pixhawk.org/peripherals/8266 say I can only do an over-the-air update if my version number is 1.0.4 or greater, but the instructions don't say how to determine what it is presently.  I blindly tried an update anyway, but it returned "FAIL".

(I am trying to diagnose a crummy connection on my factory-fresh PixRacer companion board, so updating the firmware seemed like a good place to start.)

dogmaphobic, 2016-10-17:
When using QGC to connect (through WiFi), the *WiFi Bridge* panel will show the current version. Looking through the logs, I found that OTA updates were actually implemented on version 1.0.2. If you connect to the unit using a browser to the upload "page" and actually see something asking for a firmware, you have a newer version.

While at it, this is a place for submitting bug reports. For general questions, please use gitter: [image]

closed 2016-10-17 by dogmaphobic

jpaulos, 2016-10-17:
I was intending this as a bug report on the (otherwise excellent) documentation.

Thanks for your help, I'll follow up on glitter if I can't get this to behave.  I had been looking at the QGroundControl WiFi Bridge window which doesn't have this information.  Navigating into Summary and and looking at the WifiBridge panel told me what I need (1.0.11).  I see you added this feature back in January -- thanks for that.