BeyondRobotix/mavesp8266 #34: Only integer values on http://192.168.4.1/getparameters

skleeschulte, 2016-10-22:
When navigating to http://192.168.4.1/getparameters, all the values in the list are integer values. This does not make sense e.g. for SSID and password.

I use the [pre-built firmware image v1.1.0](http://www.grubba.com/mavesp8266/firmware-1.1.0.bin) which is linked at [https://pixhawk.org/peripherals/8266#pre_built_binaries](https://pixhawk.org/peripherals/8266#pre_built_binaries).

dogmaphobic, 2016-10-22:
That's because the Mavlink protocol does not support character strings. I stash a 16 character SSID and Password into 4 32-bit (uint32_t) values each.

closed 2016-10-22 by dogmaphobic

skleeschulte, 2016-10-22:
But wouldn't it be possible to convert the the uint32_t  values back to a string?

dogmaphobic, 2016-10-22:
That's how it's done. That's how the UI in QGC handles it. The web interface is, so far, mostly a debugging tool.