BeyondRobotix/mavesp8266 #38: Strange behavior mavesp on Pixhawk

Puav, 2017-02-12:
Hi guys,
i'm facing a very strange behavior.
I'm using a NodeMCU v2 instead of an ESP-01 and it is working quite fine with an Ardupilot at 115200.
The strange happens when i try to use it with a Pixhawk on telemetry port 1: the wifi works fine and i can connect every device, but there is no communication between Pixhawk and mavesp8266.
It's worth a mention that Telem1 on Pixhawk is working fine, when i connect it to my laptop through an FTDI adapter.
I double checked all the connection, i tried different baud rate (of course both on Pixhawk and mavesp8266), and i even tried to complile my own firmware dedicated for ESP-12e, but i still cannot find a way to fix it. 
Very strange! Isn't it?

Any help will be very appreciated!
Thanks a lot in advance guys! ;)

Saxin, 2017-02-14:
usually those things end up in some simple wiring issue, please double check wiring , also try different serial ports on Pix. despite the fact that Tel. 1 is working.

michaelkoetter, 2017-02-28:
I'm seeing the same with a Wemos D1 mini - the Pixhawk Telemetry port seems fine (tested with FTDI and Bluetooth) but the ESP doesn't receive or send anything. Tried different ports and baud rates as well.
What's really weird is that it actually works on a cheap Pixhawk clone, but not on the authentic one (I really want it to work on the authentic one as I don't trust the clone for flight!)

I'm suspecting there might be some subtle difference in the signal, maybe voltage... I'll try to debug more and post my findings.

Saxin, 2017-03-01:
I can confirm that generic ESP07 / ESP12E / ESP12F modules by AI work on **clone** Pixhawk Lite and Pixhawk 2.4.8, both on 57600 and 115200 bauds.

michaelkoetter, 2017-03-03:
I built a simple board using an ESP-07 and it works with the authentic Pixhawk.
Still not sure why the WeMos D1 didn't work, maybe the onboard USB/Serial is interfering somehow?
Anyway, my custom board seems to work like a charm so I'm happy 👍

TimelessNL, 2018-04-19:
I also noticed an issue with a NodeMCU and a Ardupilot2.8 (from ebay, so probably a clone). I could not get any serial communication through. I even tried EspLink and other transparent TCP-->UART software and neither of them worked.

Personally I think this is because the ESP need's a UART level shifter to make it interface with the ArduPilot. I do have a dedicated ESP-07 with level shifter that works flawlessly. The ESP should be 5v tolerant on the I/O's but maybe the ArduPilot is not registering the ESP's 3.3v as a logic HIGH.

batzkass, 2018-04-21:
Hi,
I had the exact same problem and I have a kind of fix for it. The usb interface of this kind of NodeMCU boards can "interfere" with the ESP8266 Tx/Rx, even though the usb is disconnected. So in my WeMos D1 Mini clone, I have cut the Tx and Rx traces on the back of the board so that Tx and Rx wires only go to the ESP8266 Tx and Rx (and no longer to the USB interface). Now it works, BUT obviously you can't flash with USB anymore, unless you temporarily re-solder the traces you have cut. Flashing via WiFi remains the better option.
You can find attached photos of my modified board.
[image]

TimelessNL, 2018-04-21:
I could give that a try. Strange that there are people having succes with the NodeMCU without modifications.

Seeing your picture you could also change the resistor to a jumper so USB flashing becomes posible after connecting the jumpers.

GlidLov, 2018-08-02:
Same here, had some nodemcu which never worked with mavesp until i removed the USB uart interface chip with hot air gun, then they started to work. Should be mentioned clearly in documentation, until Gus find a sw workaround [if possible]

miniscientist, 2020-03-07:
> Hi,
> I had the exact same problem and I have a kind of fix for it. The usb interface of this kind of NodeMCU boards can "interfere" with the ESP8266 Tx/Rx, even though the usb is disconnected. So in [...]

thank you so much i racked my brains for 2 days now i found this post... thanks it worked !!!!