BeyondRobotix/mavesp8266 #40: Versioning / pre-built binaries

skleeschulte, 2017-03-23:
On the [pixhawk.org ESP8266 WiFi module setup page](https://pixhawk.org/peripherals/8266) there is [a link to this repository](https://pixhawk.org/peripherals/8266#build_from_sources) as well as [a link to a pre-built binary](https://pixhawk.org/peripherals/8266#pre_built_binaries). The version of the pre-built binary is 1.1.0, but there are no tags or releases in this repository, and neither can I find any version information in commit messages.

Where does the version number of that pre-built binary come from and what commit is it equal to? Version 1.1.0 has been linked on pixhawk.org at least since october 2016, and the latest commit here is from 02/02/2017, so it cannot be the latest comit.

Will there be any updates to the pre-built binary?

dogmaphobic, 2017-03-23:
I don't have the time to keep track of all things. It's just me here. I bump the version each time I send and build a PR (and update the links above). The last build is from Sep 2016. The PR from Feb 2017 has not been built or distributed.

dogmaphobic, 2017-03-27:
I've updated the current build (1.1.1) and added a link to the main page.

closed 2017-03-27 by dogmaphobic