BeyondRobotix/mavesp8266 #41: MAVLink 2.0 support

LorenzMeier, 2017-05-20:
More and more MAVLink functions require 2.0 support, such as e.g. Geofence transfer. It would be great to have this here too.

wolkesson, 2018-02-14:
Is anyone working on MAVLink 2 support? Seems to be fairly easy to upgrade the lib to the newer version. The problem now is that the latest firmware for PixRacer and QCS both assume Mavlink 2 but can't connect because the ESP module, that is oftentimes delivered with the PixRacer, does not support it.

dogmaphobic, 2018-02-15:
It's been in the todo list for ever. It's not as simple as it sounds. Simple would be to make it "fixed" to V2 (not support V1). To support both it's a different story. With that said, I have not had the time to look at any of this.