BeyondRobotix/mavesp8266 #42: Multiple UAVs over WiFi?

jonathanrutherford, 2017-06-06:
Hi guys, thanks for your awesome work!
I'm trying to connect a couple of pixracer drones to qgroundcontrol to test RTK, is there any way to set up the esp8266 so I can see both? I've tried to set them up as STA's but I can't connect to them from qgc like this. Any clues?
Again, many thanks for your great effort,

Jono

dogmaphobic, 2017-06-06:
These questions are better asked here: https://gitter.im/dogmaphobic/mavesp8266

If you want to set them up in *Station* mode, you will need an AP somewhere (a Wi-Fi router) so all nodes involved can connect to, including QGC. That way, they are all part of the same network.

closed 2017-06-06 by dogmaphobic

jonathanrutherford, 2017-06-07:
Ok, thanks.