BeyondRobotix/mavesp8266 #45: Change to c_library_v2 (v2.0) submodule version of Mavlink?

samacumen, 2017-08-15:
Currently, the mavesp8266 library takes support of older version of Mavlink. The problem is that if a custom ground control software uses Mavlink v2.0, it needs to detect the older version and change its own mavlink version in real-time.

Most of today's custom software uses Mavlink v2.0. Could you please look into switching to the later version and compatibility ?

PS: 
The only way to currently switch to other version of mavlink (to avoid checksum error when sending/receiving) is by toggling the mavlink status flag, which might confuse other users.

    mavlink_status_t* mavlinkStatus = mavlink_get_channel_status(mavlink_channel);
    mavlinkStatus->flags |= MAVLINK_STATUS_FLAG_OUT_MAVLINK1;

Thanks!
Shyam

tve, 2017-10-29:
Anyone in the know care to reply? I'm also confused about this...

LorenzMeier, 2017-10-29:
The MAVLink version detection docs are here:
https://mavlink.io/en/mavlink-version.html

I don't think the ESP Firmware implements this yet. I'm sure @dogmaphobic would appreciate a pull request.