BeyondRobotix/mavesp8266 #51: Heartbeat..? uploaded firmware but it is not work......

Seunghwan17, 2018-04-05:
Finally, I connect a pixracer, then I can get a parameters and status (using 192.168.4.2/getstatus).

But, when I use QGC, it can not be connected. And I used mavproxy that show heartbeat waiting....

I really don't know what is the happen and what is the matter...

Anybody help me this problem...?

mjs513, 2018-06-18:
I have a the same problem - I can not connect to QGC but it seems to work with APM planner.  Have you tried APM Planner as a test?  Have you tried to add connections manually?

Seunghwan17, 2018-06-19:
In my case, it is my mistake that wiring problem. Could show that APM and QGC? I finally connect with QGC , if can I help you I will help you.

siddhantTech, 2022-08-12:
Hi

> In my case, it is my mistake that wiring problem. Could show that APM and QGC? I finally connect with QGC , if can I help you I will help you.

Hi [Seunghwan17]
I  have been facing the same problem under the below configuration:

- Matek -H743-mini as the FC (Flight controller) with PX4 firmware for Autopilot
- ESP8266 with mavesp8266 as the firmware from dogmaphobic/mavesp8266
mavesp8266 as the firmware from dogmaphobic/mavesp8266

With this set-up in place, the laptop can connect with Wifi SSID pixracer but mavesp8266 is not able to get connected to matek-H743-mini firmware on below UART port

-  TX1, RX1 (Serial no 2 or telem2) 
- TX7, RX7 (Serial no. 1 or telem1) 
as attached the UART mapping diagram of matek FC.
[image]

Below is output from mavlink status on nsh:
 For below output, the wire is connected UART- TX1 and RX1
instance #0:
    mavlink chan: #0
    type:        GENERIC LINK OR RADIO
    flow control: OFF
    rates:
      tx: 445.3 B/s
      txerr: 0.0 B/s
      tx rate mult: 0.412
      tx rate max: 1200 B/s
      rx: 0.0 B/s
      rx loss: 0.0%
    FTP enabled: YES, TX enabled: YES
    mode: Normal
    MAVLink version: 2
    transport protocol: serial (/dev/ttyS2 @921600)

instance #1:
    GCS heartbeat valid
    mavlink chan: #1
    type:        USB CDC
    flow control: ON
    rates:
      tx: 20764.4 B/s
      txerr: 0.0 B/s
      tx rate mult: 1.000
      tx rate max: 100000 B/s
      rx: 47.0 B/s
      rx loss: 0.0%
    Received Messages:
      sysid:255, compid:190, Total: 44 (lost: 0)
    FTP enabled: YES, TX enabled: YES
    mode: Config
    MAVLink version: 2
    transport protocol: serial (/dev/ttyACM0 @2000000)
    ping statistics:
      last: 0.76 ms
      mean: 4.86 ms
      max: 82.09 ms
      min: 0.76 ms
      dropped packets: 0

Debug log from ESP8266.

........................................Got 1 client(s)
Start WiFi Bridge
Local IP: 192.168.4.1
Swapping uarts from TXRX to TXD2/RXD2...
Setup Finished.. Going into main loop...
Calling GCS.readMessage()
Inside GCS._readMessage() - udp_count: 0
Calling Vehicle.readMessage()
Inside Vehicle._readMessage() if Serial Available
After Serial Read
Calling GCS.readMessage()
Inside GCS._readMessage() - udp_count: 0 
Calling Vehicle.readMessage()
Calling GCS.readMessage()

Please let me know for any solution to this problem.

Thanks 
Sidd Tech
With this set-up in place, the laptop can connect with Wifi SSID pixracer but mavesp8266 is not able to get connected to matek-H743-mini firmware on below UART port

-  TX1, RX1 (Serial no 2 or telem2) 
- TX7, RX7 (Serial no. 1 or telem1) 
as attached the UART mapping diagram of matek FC.
[image]

Below is output from mavlink status on nsh:
 For below output, the wire is connected UART- TX1 and RX1
instance #0:
    mavlink chan: #0
    type:        GENERIC LINK OR RADIO
    flow control: OFF
    rates:
      tx: 445.3 B/s
      txerr: 0.0 B/s
      tx rate mult: 0.412
      tx rate max: 1200 B/s
      rx: 0.0 B/s
      rx loss: 0.0%
    FTP enabled: YES, TX enabled: YES
    mode: Normal
    MAVLink version: 2
    transport protocol: serial (/dev/ttyS2 @921600)

instance #1:
    GCS heartbeat valid
    mavlink chan: #1
    type:        USB CDC
    flow control: ON
    rates:
      tx: 20764.4 B/s
      txerr: 0.0 B/s
      tx rate mult: 1.000
      tx rate max: 100000 B/s
      rx: 47.0 B/s
      rx loss: 0.0%
    Received Messages:
      sysid:255, compid:190, Total: 44 (lost: 0)
    FTP enabled: YES, TX enabled: YES
    mode: Config
    MAVLink version: 2
    transport protocol: serial (/dev/ttyACM0 @2000000)
    ping statistics:
      last: 0.76 ms
      mean: 4.86 ms
      max: 82.09 ms
      min: 0.76 ms
      dropped packets: 0

Debug log from ESP8266.

........................................Got 1 client(s)
Start WiFi Bridge
Local IP: 192.168.4.1
Swapping uarts from TXRX to TXD2/RXD2...
Setup Finished.. Going into main loop...
Calling GCS.readMessage()
Inside GCS._readMessage() - udp_count: 0
Calling Vehicle.readMessage()
Inside Vehicle._readMessage() if Serial Available
After Serial Read
Calling GCS.readMessage()
Inside GCS._readMessage() - udp_count: 0 
Calling Vehicle.readMessage()
Calling GCS.readMessage()

Please let me know for any solution to this problem.

Thanks 
Sidd Tech