BeyondRobotix/mavesp8266 #56: Question on Recieving Messages to GCS

mjs513, 2018-05-27:
Hi dogmaphobic
I came across you library as I was trying to interface an Teensy 3.5 MCU to an ESP8266 to go to a GCS.  I was fairly successful with just using straight UDP to get flight data (GPS/IMU) messages to the APM planner but was having difficulties with going the other way (GCS -> Vehicle).  So I decided to give this a try.  

I uploaded this via the Arduino IDE (yes I converted main into an INO file) and that worked fine.  It took me a little while to get it working but I was successful in getting the GCS to read messages from the Arduino to bridge to the GCS.  It works like a charm.  It connects right up.  

In Wireshark I can see messages going from the GCS to the vehicle but am having problems figuring out how to actually read them on the Arduino side.  Probably because I don't fully understand the telemetry protocol.

Right now I am writing mavlink messages to Serial1 which is working fine.  But if I just try and read Serial1 does really work.  So the question is what is the discrimator to know when receiving a message or sending a message.  Is there a hardware connection I am missing between the ESP8266 and the Arduino for flow control??

Thanks
Mike

dogmaphobic, 2018-05-27:
> Right now I am writing mavlink messages to Serial1 which is working fine.

Does that mean the GCS is receiving the MAVLink messages?

>But if I just try and read Serial1 does really work.

There could so many reasons, I don't even know where to begin. Either way, this is not the right place for usage questions. You will have better luck here: [image]

closed 2018-05-27 by dogmaphobic

mjs513, 2018-05-27:
> Does that mean the GCS is receiving the MAVLink messages?
Yes the GCS is receiving the messages fine, no problem.

> this is not the right place for usage questions. 
Didn't know about glitter - I will post the same message over there and see if I can figure it out.

Thank you for the response.
Mike