BeyondRobotix/mavesp8266 #66: v1.2.2 and v1.1.1 fails to bootup (Stuck in bootloader ? )

echoGee, 2019-05-31:
Unable to see the pixracer wifi after flashing the firmware v1.2.2 or v1.1.1 . Steps to reproduce 
1. Use the hardware: [ESP8266EX](https://www.sparkfun.com/products/13678) along with [USB-TTL convertor](https://www.amazon.com/gp/product/B078W5L8W1/ref=ppx_yo_dt_b_asin_title_o02_s00?ie=UTF8&psc=1)
2. Follow https://docs.px4.io/en/telemetry/esp8266_wifi_module.html#flashing-the-esp8266-firmware to upload the firmware.
ESP is powered using a power supply rated for multiple amps of current
**Flash mode**:
CH_PD is pulled high 
GPIO0 is connected to ground
**Normal mode**:
Disconnect Ch_PD and GPIO0
3. Power cycle ESP8266 and check for wifi connections 
4. Read the Serial output(at 78400 baud) to get the following output(ESP still in bootloader ?): 

>  ets Jan  8 2013,rst cause:1, boot mode:(3,2)

>load 0x4010f000, len 1384, room 16 
>tail 8
chksum 0x2d
csum 0x2d
v8b899c12
~ld

[image]
[image]

Any suggestions to solve this ?

dogmaphobic, 2019-05-31:
Given the code hasn't changed in well over a year, my guess would a defective ESP8266 module? Have you tried flashing something else just to test it?

echoGee, 2019-05-31:
@dogmaphobic : Yes I have and it did work. I also tried 3 different ESP8266 modules

Bumerangas, 2019-06-07:
for me too not flashing, stuck at some point at mid with nodemcu-flasher-master tool.

i was able to flash it ~half year ago and now bricking my wifi module this FW and same situation with latest firmware-esp01_1m.bin after flash - module brick, especially via web interface

Bumerangas, 2019-06-07:
this problem started after i updated my esp-01 module to this FW
http://firmware.ardupilot.org/Tools/MAVESP8266/2019-01-23/

echoGee, 2019-06-07:
> for me too not flashing, stuck at some point at mid with nodemcu-flasher-master tool.
> 
> i was able to flash it ~half year ago and now bricking my wifi module this FW and same situation with latest firmware-esp01_1m.bin after flash - module brick, especially via web interface

This wasn't the issue I had though

Bumerangas, 2019-06-07:
Solved after changing FTDI adapter 5V to 3.3V.