BeyondRobotix/mavesp8266 #67: Could not visit 192.168.4.1

dayL-W, 2019-06-01:
I could not visit 192.168.4.1 when i connect to "Pixracer" WiFi.
I ensure that my computer has connect the "Pixracer" and get IP "192.168.4.2".

I use the firmware build from the code.
I use Ubuntu 18.04
My board is ESP01
My use "platformio run -e esp01 -t upload" to uplaod

who could help me?

dayL-W, 2019-06-01:
I could ping 192.168.4.1

dayL-W, 2019-06-01:
When I uplaod Firmware with Firmware version 1.2.2, there is no problem.
Is there any different between Firmware version 1.2.2 and .pioenvs/esp01_1m/firmware.bin ?

dayL-W, 2019-06-02:
There must be a different between Firmware version 1.2.2 and .pioenvs/esp01_1m/firmware.bin, even the size is different

simopanzer89, 2020-04-05:
Same problem here.
@dayL-W did you find a solution for this?