BeyondRobotix/mavesp8266 #68: How to build in windows10?

hsy77101, 2019-07-28:
Hello.
I checked all the texts.
But I don't understand how to run the following in Windows 10.
Do I need to run the following commands in the only Linux like ubuntu?
"
brew install platformio
git clone --recursive https://github.com/dogmaphobic/mavesp8266.git
cd mavesp8266
platformio run
"

dogmaphobic, 2019-07-28:
Those are instruction for macOS. Go here for instructions on how to install Platformio on other platforms: https://platformio.org/install

closed 2019-07-28 by dogmaphobic

skleeschulte, 2021-05-11:
What I just found out when trying to compile version 1.2.2 with PlatformIO in VS Code on Windows 10: the `build_flag` setting in the platformio.ini config file uses the external command `echo`, which works different on Windows:

    build_flags = !echo "-DPIO_SRC_REV="$(git rev-parse HEAD) "-DPIO_BUILD_DATE="$(date +%%Y-%%m-%%d) "-DPIO_BUILD_TIME="$(date +%%H:%%M:%%S) ${version_env.version_flags}

As a quick fix, I set the variables manually:

    #build_flags = !echo "-DPIO_SRC_REV="$(git rev-parse HEAD) "-DPIO_BUILD_DATE="$(date +%%Y-%%m-%%d) "-DPIO_BUILD_TIME="$(date +%%H:%%M:%%S) ${version_env.version_flags}
    build_flags = "-DPIO_SRC_REV=3c8c130ce2f0865de3469b1648453a695af2c948" "-DPIO_BUILD_DATE=2021-05-11" "-DPIO_BUILD_TIME=14:44:48" ${version_env.version_flags}

This way, I can successfully run `pio run` in a terminal window in the PlatformIO-IDE. Note that the `pio` command is only available when opening a new terminal windows in the IDE, it does not work in the default powershell terminal window.