BeyondRobotix/mavesp8266 #76: F not declared in this scope

ChrisWighton, 2020-04-01:
Hi guys, working on a new project using the ESP8266 for a small wifi module. When trying to build the project we get this error "F not declared in this scope". 
We know this is an Arduino function, not something written in on our project but can anyone point us in the right direction as to where this error might be coming from and how to fix it. 
We have tried the old turn it off and on again trick and updated Arduino and its libraries already but no success. One computer it works, the other it does not. 

Any help would be greatly appreciated.

d-skinner, 2020-05-01:
I'm having the same problem.

`In file included from .platformio/packages/framework-arduinoespressif8266/libraries/ESP8266WebServer/src/ESP
8266WebServer.h:235:0, from src/mavesp8266_httpd.cpp:44:
.platformio/packages/framework-arduinoespressif8266/libraries/ESP8266WebServer/src/ESP8266WebServer-impl.h: 
In member function 'bool esp8266webserver::ESP8266WebServerTemplate<ServerType>::authenticate(const char*, const char*)':
.platformio/packages/framework-arduinoespressif8266/libraries/ESP8266WebServer/src/ESP8266WebServer-impl.h:1
25:36: error: there are no arguments to 'F' that depend on a template parameter, so a declaration of 'F' must be available [-fpermissive]
     if(authReq.startsWith(F("Basic"))){`

Is the first error I receive

guglie, 2020-05-12:
You can use this fix (currently in master) https://github.com/dogmaphobic/mavesp8266/commit/f4413b53168faedef6814ab51de84b533d121205

basically the following include has to be first

https://github.com/dogmaphobic/mavesp8266/blob/f4413b53168faedef6814ab51de84b533d121205/src/mavesp8266_httpd.cpp#L38