BeyondRobotix/mavesp8266 #84: Compiling error

Owiec94, 2021-04-02:
Dear
I've error during compiling Your solution

> Executing task: C:\Users\Admin\.platformio\penv\Scripts\platformio.exe run --environment esp12e <

Processing esp12e (platform: espressif8266; framework: arduino; board: esp12e)
--------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------

Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/esp12e.html
PLATFORM: Espressif 8266 (2.3.0) > Espressif ESP8266 ESP-12E
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
PACKAGES:
 - framework-arduinoespressif8266 2.20601.191124 (2.6.1)    
 - tool-esptool 1.413.0 (4.13)
 - tool-esptoolpy 1.20800.0 (2.8.0)
 - toolchain-xtensa 2.40802.200502 (4.8.2)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 30 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <ESP8266mDNS> 1.2
|   |-- <ESP8266WiFi> 1.0
|-- <ESP8266WiFi> 1.0
|-- <ESP8266WebServer> 1.0
|   |-- <ESP8266WiFi> 1.0
|-- <EEPROM> 1.0
Building in release mode
Compiling .pio\build\esp12e\src\main.cpp.o
Compiling .pio\build\esp12e\src\mavesp8266.cpp.o
Compiling .pio\build\esp12e\src\mavesp8266_component.cpp.o
Compiling .pio\build\esp12e\src\mavesp8266_gcs.cpp.o
xtensa-lx106-elf-g++: error: +%Y-%m-%d): No such file or directory
xtensa-lx106-elf-g++: error: +%H:%M:%S): Invalid argument
xtensa-lx106-elf-g++: error: +%Y-%m-%d): No such file or directory
xtensa-lx106-elf-g++: error: +%H:%M:%S): Invalid argument
*** [.pio\build\esp12e\src\mavesp8266_component.cpp.o] Error 1
*** [.pio\build\esp12e\src\main.cpp.o] Error 1
xtensa-lx106-elf-g++: error: +%Y-%m-%d): No such file or directory
xtensa-lx106-elf-g++: error: +%H:%M:%S): Invalid argument
xtensa-lx106-elf-g++: error: +%Y-%m-%d): No such file or directory
xtensa-lx106-elf-g++: error: +%H:%M:%S): Invalid argument
*** [.pio\build\esp12e\src\mavesp8266.cpp.o] Error 1
*** [.pio\build\esp12e\src\mavesp8266_gcs.cpp.o] Error 1

Can You help me?

FAIZALR, 2021-05-11:
Even i face same issue...any solution?

Owiec94, 2021-05-11:
> Even i face same issue...any solution?

So far no, I've successfully compile this solution under Arduino. Wait for the answer of @dogmaphobic

FAIZALR, 2021-05-18:
Hi,I have solved the issue...it's working now perfectly :)
In case any further clarification..u can contact my WhatsApp number +919449228982
-Faiz

tkhamvilai, 2021-11-08:
I got the same error. Any solutions?